embassy-rs/lora-phy #2: Allow for different TCXO control voltages for 126X

dberlin, 2023-03-11:
The TCXO control voltage is hard coded in set_oscillator to be 1.7V, but it's normally configurable because different radios use different voltages.

It would be nice to have it controllable.

I'm happy to make a PR if you tell me how you want it to work :)

ceekdee, 2023-03-11:
Your query is timely.  I will be making updates this weekend to focus on board type (Stm32wlSx1262, Stm32l0Sx1276, and Rak4631Sx1262 are the initial ones) rather than chip type (Sx1261, Sx1262, Sx1276, etc.) within a radio kind (Sx1261/2, Sx1276/7/8/9).  This will allow us to better account for differences in LoRa boards even with the same LoRa chip.  As we add more LoRa boards to the mix, we will discover variations on a theme that can be captured in a better manner than adding a bunch of board type checks.

ceekdee, 2023-03-11:
There is a LoRa/LoRaWAN group chat using Element:

https://app.element.io/#/home

It is currently not public, but I would be glad to invite you if you have (or create) an Element account and let me know the name of the account.

ceekdee, 2023-03-11:
The latest update supports board type (and within board type, chip type).

```
pub enum BoardType {
    GenericSx1261, // placeholder for Sx1261-specific features
    Rak4631Sx1262,
    Stm32l0Sx1276,
    Stm32wlSx1262,
}

pub enum ChipType {
    Sx1261,
    Sx1262,
    Sx1276,
    Sx1277,
    Sx1278,
    Sx1279,
}

impl From<BoardType> for ChipType {
    fn from(board_type: BoardType) -> Self {
        match board_type {
            BoardType::GenericSx1261 => ChipType::Sx1261,
            BoardType::Rak4631Sx1262 => ChipType::Sx1262,
            BoardType::Stm32l0Sx1276 => ChipType::Sx1276,
            BoardType::Stm32wlSx1262 => ChipType::Sx1262,
        }
    }
}
```

which prevents generic chip type checks from having to be modified each time a new board is added:

```
       match self.radio_kind.get_board_type().into() {
            ChipType::Sx1261 | ChipType::Sx1262 => {
                ModulationParams::new_for_sx1261_2(spreading_factor, bandwidth, coding_rate, frequency_in_hz)
            }
            ChipType::Sx1276 | ChipType::Sx1277 | ChipType::Sx1278 | ChipType::Sx1279 => {
                ModulationParams::new_for_sx1276_7_8_9(spreading_factor, bandwidth, coding_rate, frequency_in_hz)
            }
        }
```
        
while making true board type checks evident:

```
    async fn init_rf_switch(&mut self) -> Result<(), RadioError> {
        if self.board_type != BoardType::Stm32wlSx1262 {
            let op_code_and_indicator = [OpCode::SetRFSwitchMode.value(), true as u8];
            self.intf.write(&[&op_code_and_indicator], false).await?;
        }
        Ok(())
    }
```

The chip type name is included in the board type name in case the same board type may include different chip types.

Therefore:
- in mod_params.rs, add your board type and update the From specification for the board's chip type;
- update the pertinent radio kind function as in the init_rf_switch() sample above.

dberlin, 2023-03-16:
Awesome.
I'll take a whack at it, and making the Tcxo control part of the board type this weekend

closed 2023-03-26 by ceekdee