embassy-rs/lora-phy #27: LoRa Receive Soft Lock

ProjectInitiative, 2023-09-12:
## Summary

When running [this](https://github.com/embassy-rs/embassy/blob/2c38dd17b9a5a507fc77d90d78486e4764234752/examples/rp/src/bin/lora_p2p_receive.rs#L107) example, there is a soft lock. It can be reproduced by sending another message while the match is waiting, the `rx` function get stuck, and cannot receive any more messages until the micro controller is reset.

## Hardware Summary

* x 2 [RP2040](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html) (1 tx, 1 rx)
* x 2 [SX1262 Waveshare LoRa Modules](https://www.waveshare.com/pico-lora-sx1262-868m.htm)  (1 tx, 1 rx)

## Steps to reproduce

Copy [line 89 - 98](https://github.com/embassy-rs/embassy/blob/2c38dd17b9a5a507fc77d90d78486e4764234752/examples/rp/src/bin/lora_p2p_send.rs#L89) and paste after  [line 98](https://github.com/embassy-rs/embassy/blob/2c38dd17b9a5a507fc77d90d78486e4764234752/examples/rp/src/bin/lora_p2p_send.rs#L98) in the `lora_p2p_send.rs` file.

Compile and load on the respective pico devices, and after the first message, any messages sent after are lost, and the receive loop does not continue.

<details>
  <summary>Trace Log</summary>

```bash
(HOST) INFO  flashing program (14 pages / 56.00 KiB)
(HOST) INFO  success!
(HOST) WARN  insufficient DWARF info; compile your program with `debug = 2` to enable location info
(HOST) WARN  `defmt::timestamp!` implementation was found, but timestamp is not part of the log format;
────────────────────────────────────────────────────────────────────────────────
TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ <mod path> @ <file>:0
TRACE write: 0x[c0, 0]
└─ <mod path> @ <file>:0
TRACE InputFuture::poll return Poll::Pending
└─ <mod path> @ <file>:0
TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ <mod path> @ <file>:0
TRACE write: 0x[9d, 1]
└─ <mod path> @ <file>:0
TRACE InputFuture::poll return Poll::Pending
└─ <mod path> @ <file>:0
TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ <mod path> @ <file>:0
TRACE write: 0x[80, 0]
└─ <mod path> @ <file>:0
TRACE InputFuture::poll return Poll::Pending
└─ <mod path> @ <file>:0
TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ <mod path> @ <file>:0
TRACE write: 0x[8a, 1]
└─ <mod path> @ <file>:0
TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ <mod path> @ <file>:0
TRACE write: 0x[d, 7, 40, 14, 24]
└─ <mod path> @ <file>:0
TRACE InputFuture::poll return Poll::Pending
└─ <mod path> @ <file>:0
TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ <mod path> @ <file>:0
TRACE write: 0x[97, 1, 0, 2, 80]
└─ <mod path> @ <file>:0
TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ <mod path> @ <file>:0
TRACE write: 0x[96, 1]
└─ <mod path> @ <file>:0
TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ <mod path> @ <file>:0
TRACE write: 0x[8f, 0, 0]
└─ <mod path> @ <file>:0
TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ <mod path> @ <file>:0
TRACE write: 0x[1d, 8, d8, 0]
└─ <mod path> @ <file>:0
TRACE read 1: 0x[c8]
└─ <mod path> @ <file>:0
[... 195 lines cut ...]
└─ <mod path> @ <file>:0
TRACE read 2 status 0xd4: 0x[3, 0]
└─ <mod path> @ <file>:0
TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ <mod path> @ <file>:0
TRACE write: 0x[1e, 0, 0]
└─ <mod path> @ <file>:0
TRACE read 3: 0x[1, 2, 3, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
└─ <mod path> @ <file>:0
TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ <mod path> @ <file>:0
TRACE write: 0x[14]
└─ <mod path> @ <file>:0
TRACE read 3 status 0xd4: 0x[b8, 21, ba]
└─ <mod path> @ <file>:0
INFO  rx successful
└─ <mod path> @ <file>:0
DEBUG Loop intera
└─ <mod path> @ <file>:0
DEBUG process_irq loop entered
└─ <mod path> @ <file>:0
TRACE InputFuture::poll return Poll::Pending
└─ <mod path> @ <file>:0
```
</details>

ceekdee, 2023-09-12:
Is this a debug statement you added?  I am not picking it up in embassy or lora-phy.

DEBUG Loop intera

ProjectInitiative, 2023-09-12:
Oh yes, my mistake, I added an additional debug to the example to try and see where the iteration was hanging. I added that between lines 96 and 97 of the recv example.

ceekdee, 2023-09-12:
If this is indeed the end of the trace:

TRACE read 3 status 0xd4: 0x[b8, 21, ba]
└─ <mod path> @ <file>:0
INFO  rx successful
└─ <mod path> @ <file>:0
DEBUG Loop intera
└─ <mod path> @ <file>:0
DEBUG process_irq loop entered
└─ <mod path> @ <file>:0
TRACE InputFuture::poll return Poll::Pending
└─ <mod path> @ <file>:0

it seems to indicate the polling on the DIO pin (looking for it to go high) is terminating after one poll.

Please provide the revision of Embassy in GitHub which is being used for this test.

ProjectInitiative, 2023-09-12:
I wonder if it is because I am using an old version without realizing it: https://github.com/embassy-rs/lora-phy/commit/59e9c344872a4825811b14c3e688f30e20a5deed

Going to re run tests on newest release.

I ran the examples over night twice, and that is how I discovered the issue, because at some point the receive got stuck.

ProjectInitiative, 2023-09-12:
For the sake of sanity, I cloned a clean repo of the embassy repo. Here is the stack trace:

<summary>TRACE LOG</summary>
<details>

```bash
0.042856 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.042933 TRACE write: 0x[c0, 0]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:43
0.043067 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.043096 TRACE write: 0x[9d, 1]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:43
0.043190 TRACE InputFuture::poll return Poll::Pending
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.043246 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.043271 TRACE write: 0x[80, 0]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:43
0.043364 TRACE InputFuture::poll return Poll::Pending
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.043411 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.043437 TRACE write: 0x[8a, 1]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:43
0.043560 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.043589 TRACE write: 0x[d, 7, 40, 14, 24]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:43
0.043707 TRACE InputFuture::poll return Poll::Pending
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.043753 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.043777 TRACE write: 0x[97, 1, 0, 2, 80]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:43
0.043886 TRACE InputFuture::poll return Poll::Pending
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.043919 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.043945 TRACE write: 0x[96, 1]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:43
0.044044 TRACE InputFuture::poll return Poll::Pending
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.044077 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.044100 TRACE write: 0x[8f, 0, 0]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:43
0.044407 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.044455 TRACE write: 0x[1d, 8, d8, 0]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:102
0.044496 TRACE read 1: 0x[c8]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:112
0.044630 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.044661 TRACE write: 0x[d, 8, d8, de]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:43
0.044796 TRACE InputFuture::poll return Poll::Pending
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.044850 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.044873 TRACE write: 0x[95, 4, 7, 0, 1]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:43
0.044918 DEBUG tx power = 0
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/sx1261_2/mod.rs:444
0.045001 TRACE InputFuture::poll return Poll::Pending
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
0.045034 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
[... 176 lines cut ...]
2.994064 TRACE write: 0x[13]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:167
2.994088 TRACE read 2 status 0xd4: 0x[3, 0]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:177
2.994341 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
2.994380 TRACE write: 0x[1e, 0, 0]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:102
2.994410 TRACE read 3: 0x[1, 2, 3, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:112
2.994745 TRACE LevelLow for pin 2 was cleared, return Poll::Ready
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
[... cut ...]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:167
2.994801 TRACE read 3 status 0xd4: 0x[8c, 21, 8c]
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/interface.rs:177
2.994863 INFO rx successful
└─ src/bin/lora_p2p_receive.rs:105
7.994925 DEBUG process_irq loop entered
└─ /home/kpzak/.cargo/git/checkouts/lora-phy-eb526861eb331585/1323ecc/src/sx1261_2/mod.rs:832
7.994952 TRACE InputFuture::poll return Poll::Pending
└─ /home/kpzak/development/build-software/embassy/embassy-rp/src/fmt.rs:117
```

</details>

ceekdee, 2023-09-12:
Thank you for re-creating the issue with the current Embassy revision.  The trace tells the same story:  polling for the DIO1 interrupt pin going high seems to terminate, or there is an error in handling that event, or the logging itself fails.

I'll attempt to recreate it in my testbed and debug it further.  I may need to request assistance from the Embassy team.

There is a lora-phy version 2, but there should be no difference in how it handles a read continuous operation.  Embassy will be updated to use version 2 once another Embassy dependency is updated.

ProjectInitiative, 2023-09-12:
That appears to work. going to run another overnight test!

ceekdee, 2023-09-12:
If the overnight test verifies the fix, I will make a PR for the embassy-lora crate.

ProjectInitiative, 2023-09-12:
Probably don't need an overnight test. I had it running all day, haven't seen a single error or hang. Will continue if desired.

ceekdee, 2023-09-12:
Seems to be a sufficient test.  Thank you!  I'll work on a PR for embassy-lora and note that here when it is opened.

ceekdee, 2023-09-12:
Issue closed with PR applied to Embassy.

closed 2023-09-12 by ceekdee