embassy-rs/lora-phy #30: index out of bounds: the len is 9 but the index is 9;

vDorst, 2023-09-25:
````0.361316 DEBUG tx power = 0
└─ /home/r/.cargo/registry/src/index.crates.io-6f17d22bba15001f/lora-phy-2.1.2/src/sx1261_2/mod.rs:440
0.361870 ERROR panicked at /home/r/.cargo/registry/src/index.crates.io-6f17d22bba15001f/lora-phy-2.1.2/src/sx1261_2/mod.rs:72:36:
index out of bounds: the len is 9 but the index is 9
``````

Code snippet and code crashed in `LoRa::new()`;
```
```   let iv = GenericSx126xInterfaceVariant::new(nss, reset, dio1, busy, None, None).unwrap();

    let mut lora = {
        match LoRa::new(
            SX1261_2::new(BoardType::RpPicoWaveshareSx1262, lora_spi, iv),
            false,  
            &mut delay,
        )
        .await
        {
            Ok(l) => l,
            Err(err) => {
                info!("Radio error = {}", err);
                return;
            }
        }
    };
```
But it doesn't crash if I set `enable_public_network` true.

ceekdee, 2023-09-25:
The "index out of bounds: the len is 9 but the index is 9" is an indication that the first SPI read operation attempted failed.  It is unclear to me why the enable_public_network setting should have an effect on that.

Are you running one of the Embassy examples, and what combination of MCU/LoRa board are you running?  The first thing to look at is the setup for SPI.

vDorst, 2023-09-25:
I found it, SPI frequency is way to high, 64MHz instead of the max 16 Mhz.
64 Mhz is used for the LCD which is also sharing the SPI bus.

Yes, I am using the embassy examples but I combined two of them to have LCD and LORA.
But sharing SPI bus is not so easy, It works I think but it is really ugly because LORA lib don´t accept SpiDevice.
So I had to write a wrapper around SpiBus to make it work.

This is the hardware I use.
Baseboard: https://www.waveshare.com/rp2040-lcd-0.96.htm (Modified: Moved LCD reset line to GP13)
Loraboard: https://www.waveshare.com/wiki/Pico-LoRa-SX1262

closed 2023-09-25 by vDorst

ceekdee, 2023-09-25:
This is excellent information.  lora-phy assumes ownership of the SPI bus because there is at least one MCU/LoRa chip combo that makes doing anything else problematic, if I remember correctly.  I am glad to know there is an identified need for lora-phy to share the bus in some cases and have opened an issue for future consideration - unfortunately, I will be unavailable to work on that until the end of October.

Feel free to provide additional guidance in the issue I have opened.